imbushuo/mac-precision-touchpad #562: Can't get magic trackpad 3 to work with driver

johnydazzles, 2025-04-04:
I followed the instructions and downloaded the latest release (3979) and installed using the inf file.  Then I paired the magic trackpad with bluetooth.  However I still only have basic functionality (moving the mouse and pressing to left click) which was available even before I installed the driver.

FrankLaterza, 2025-04-17:
any update on this?

CEbbinghaus, 2025-04-30:
Me 3. Cannot find the drivers in the device manager either

chrisbraddock, 2025-05-24:
Windows seems to use some built-in Microsoft driver. I've tried deleting the device and rebooting and is just recognized now and I can't seem to override the driver.

vbrz, 2025-09-03:
Support for MT3 seems to have been committed after the latest release. A fresh build might solve this?

konradzuse, 2025-10-22:
Did you get anywhere with BT on the MT3 with device id VID&0001004c_PID&0324 ?

I have usb working (arm64 Surface Pro 11) but can't get the bluetooth driver to install.