imbushuo/mac-precision-touchpad #565: Doesn't provide multi touches capabilities as expected

naivechou, 2025-05-14:
**Pre-report checks**

- [x] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?
- I did. It seems that there are some similar issues
	- [User Story 759 Windows11可以使用嘛？ (GitHub Issue #433)](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/edit/759/)
	- [User Story 744 Magic Mouse driver? (GitHub Issue #424)](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/edit/744/)
	- [User Story 832 Confirmed working on Windows 11 (GitHub Issue #481)](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/edit/832/)
	- [User Story 808 I can´t use magic trackpad 2 with bluetooth on Windows 10 (GitHub Issue #467)](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/edit/808/)

**Describe the bug**
After installed the driver, the device was _successfully paired_, and _no any error reported_. But the trackpad **only act as a simple mouse**. No any multi touches detected, **not even the single tap to click**, scroll or content menu behavior with 2 fingers, not mention to the 3 or 4 fingers one.

**To Reproduce**
Installed the driver and probe the trackpad in Bluetooth seeking dialog and connect the device.

**Expected behavior**
Even without any further configurations for multi touch gestures, the system should generate default gestures behaviors with 2+ fingers.

**Traces**
The device manager information provided by the driver:

[image]

The Magic Trackpad is under the `<machine>/Bluetooth/Magic Trackpad`

In the Bluetooth section in Settings of Windows 11, the device seems to be recognized as a mouse:

[image]

**Environment**

The system information:

> Edition	Windows 11 Enterprise
> Version	24H2
> Installed on	‎1/‎22/‎2025
> OS build	26100.3775
> Experience	Windows Feature Experience Pack 1000.26100.66.0

The device model name and number is { Apple Magic Trackpad 2 model: **A3120** } . It has the USB type C interface to connect with the host, but now I'm connecting it via Bluetooth interface.

sjain882, 2025-06-30:
The readme for this repo states that Bluetooth isn't supported, but I'm not sure if thats up-to-date info.

There is this fork, however: https://github.com/1Revenger1/mac-precision-touchpad

I noticed in the commits here that there's "Fix Bluetooth" twice.

Has anyone tried this driver? Just out of curiousity. Even if it works, you'd have to disable driver signature enforcement, as only this repo provides signed drivers... but it could still be interesting to try

naivechou, 2025-07-01:
I'm now using the [Magic Utilities](https://magicutilities.net/) to solve **EVERY** problem this repo hadn't solved. I think this issue can be closed now.

sjain882, 2025-07-01:
I disagree, the problem is still present within this repository's code, so this unsolved issue remains relevant. Furthermore, that utility is payware (subscription!) versus this freeware repository.